oxidecomputer/quartz #400: DIMM temperature flicker on Cosmo

mkeeter, 2025-07-24:
@labbott and I noticed some weirdness in DIMM temperature readings:

https://github.com/user-attachments/assets/7240e7bf-d765-4181-9fb9-dd757ce60d05

This corresponds to intermittent `TemperatureReadTimeout` logs in the `cosmo-spd` ringbuf, which means that `bus0/1_rx_byte_count` didn't hit 2 bytes within 8 ms of polling ([source](https://github.com/oxidecomputer/hubris/blob/7584e1d0eeef1081539d3de1385f53c33a21b716/task/cosmo-spd/src/main.rs#L241-L262))

The behavior is weirdly bistable: we'll have solid readings from all sensors for a while, then they'll flicker wildly for a time, then go back to being stable.

It's unclear if this is a Hubris or FPGA issue, but I'm _slightly_ more suspicious of the FPGA.

nathanaelhuffman, 2025-07-30:
I'm going to need to get more DIMMs on hand. On my cosmo, with only 2 dimms installed (one on each bus) I don't see  this behavior, but the two dimm case is the least interesting. I'll get  some more dimms on hand and re-test.

nathanaelhuffman, 2025-11-12:
with a full 16 dimms installed, and running with saleae the "flicker" corresponds to SP5 transactions which take priority. I think there's a bug here on the FPGA side where the transaction is dropped or aborted without notice and causes the flicker. Everything stabilizes once SP5 dimm transactions cease.  I don't see the transactions ending up on the wire as 0's so they're likey being dropped due to arb loss by the FPGA with no reporting.

This is 100% correllated to SP5 traffic though, we see the flicker when there is SP5 i2c traffic and no flicker when there isn't. I'll dig into this more once we sort out the training and SPD cache issues.

nathanaelhuffman, 2025-12-17:
Related to #https://github.com/oxidecomputer/quartz/issues/434